EtteGit/EnragedRabbitProject #4: is this based off of SMUff?

icebal, 2021-03-19:
This project has many similarities to this project:
https://www.thingiverse.com/thing:3431438
Is this coincidence or is EnragedRabbit based on smuff?

EtteGit, 2021-03-19:
Hello. It's neither a coincidence or based on Smuff directly (i.e. it's not a fork, all was designed from scratch). I took what I consider good ideas from mmu systems (mmu2 and smuff) and added mines, resulting in the enraged rabbit project.

I guess this "issue" can be closed? Lol

icebal, 2021-03-19:
So its design is based on mmu2 and smuff, but redesigned. since inspiration and design aspects came from other projects, can a small credit be added to reflect that?

EtteGit, 2021-03-19:
Sure thing

icebal, 2021-03-19:
Woot thanks!

closed 2021-03-19 by icebal