EtteGit/EnragedRabbitProject #8: [Question] Tolerances on printed parts

GMagician, 2021-03-26:
Sorry to contact you here, I read parts are expected to be printed in abs, is there any possibility to see in the future a petg printable version?
I tryed to print something in petg but 0.1mm (filament top cap) is a problem (maybe my printer but in petg i'm used with 0.2 tolerances)

Please redirect me to a better place to be updated about this project.
Thanks

EtteGit, 2021-03-28:
Hi! It is not foreseen to make a "petg" version of the Enraged Rabbit project, but you have access to all the CAD files (steps), so if you want to adjust some parts, you're totally free of doing so =)

You can discuss this project on the Voron Discord (and you can PM me on Discord).

Cheers,

Ette

GMagician, 2021-03-28:
Thanks for you answer. I'm not confortable with step files (Fusion360 has no history on what you do) but I will try that route.
As I can see steps are "preliminary", do you think to "consolidate" this project soon? Since step are not easy to handle I would not like to do job twice.

Greetings

closed 2021-03-28 by GMagician

NameOfTheDragon, 2021-05-19:
> I'm not confortable with step files (Fusion360 has no history on what you do)

Hence #23 :)