EtteGit/EnragedRabbitProject #23: [Request] Please include *.f3d Fusion 360 CAD models if possible (as well as the STEP files)

NameOfTheDragon, 2021-05-19:
I know you've included STEP files for all the CAD models, which is great. It allows easy import into Fusion 360 for viewing the model.

However, the problem with STEP files is that they do not capture the design history, so you end up with an object with no access to the sketches, features and operations that went into creating it. Don't get me wrong, that's better than a poke in the eye, but it does make modding much more difficult than it strictly needs to be. It is so much easier if you can go back and tweak a sketch or edit a feature, or insert new operations in the timeline to modify existing features. All this is lost when exporting as a STEP file.

I don't know that you were working in Fusion 360 of course, but it's an educated guess that you likely were.

So my request is - if possible - and if you are willing - please include the *.f3d Fusion 360 files as well as the step files.

EtteGit, 2021-05-22:
Hello Tim,

While I understand your request, I clearly don't feel confortable enough sharing the parts history. Right know, in the rush of releasing it, many parts have been "hotfixed" on the go. While I wouldn't mind sharing the history on a clean part, we're not there yet.

If this need is to tweak tolerances for a print in PETG, I strongly advise, still, to use ABS. That way you'll end up with a similar setup as others, making possible troubleshooting much less painful.

Now if you want to do a "small" mod for a part or two, I can send to, by PM, the f3d for the concerned parts only.

Best regards,

Ette

syscod3, 2021-08-31:
If that is the case, why the project is not clearly marked as in alpha stage for the benefit of anyone?

EtteGit, 2021-08-31:
@syscod3 Well, a "dirty" cad history doesn't mean a dirty//bad CAD file...

Also, already on the front page since day first :
"You'll find more informations about each of those components into their respective folders. Note that this is a work in progress !!"

Mynasru, 2021-10-04:
@EtteGit Maybe an in-between solution would be to create a "dirty" branch in git so people can enjoy both your proper and hot fixed work 😃

closed 2022-06-17 by EtteGit