EtteGit/EnragedRabbitProject #38: Carrot_Feeder_V1.1.step fails to import/translate in Fusion360

loban4, 2021-09-20:
The Step File for the Carrot Feeder Assembly fails to import/translate to Fusion 360.
Unfortunately the report is rather vague: 
[Carrot_Feeder_V1.1_errorReport.pdf](https://github.com/EtteGit/EnragedRabbitProject/files/7199108/Carrot_Feeder_V1.1_errorReport.pdf)

Feel free to delete this Issue if it is outside the scope of this project.

EtteGit, 2021-09-21:
I'm not sure how you're doing it. I just downloaded the step file from the repo, and could open it in Fusion360 no problem.

I'm doing 'file -> open -> open from my computer'. And it just works.

Maybe you need to update // restart fusion?

loban4, 2021-09-21:
Problem solved.
I did the right click --> save as... To download the file, which apperantly corrupted it (it did work with the ERCP though).
Downloading it with a plugin or cloning the git solved the problem.