EtteGit/EnragedRabbitProject #50: ERCF calibrate failing

lerou114, 2021-11-13:
I have successful run all my calibrations and I am on the step to calibrate the ercf when I run the calibration I have set the bowden length in config 5 cm shorter than length when I run the ERCF_CALIBRATE command when I get to the part where the filament feeds to the filament to the extruder I have klipper shutting down every time.

[image]

[image]

I reran the encoder calibration and I get exact 200 mm measurements multiple times during encoder calbration step. I'm at a lose as to what I am doing incorrectly. What should I be looking for?

EtteGit, 2021-11-13:
Make sure the klipper version on your MCUs matches the one on your rpi

lerou114, 2021-11-13:
I just checked  there was a version mismatch so I have flashed everything to the newest version of klipper, but I am still getting the error.  Currently this is my version:
[image]

EtteGit, 2021-11-13:
Ok, thanks.  Can you provide the klippy.log when this error occurs please?

lerou114, 2021-11-13:
So I cleared my log file and recreated the condition I happened to find this in the log at line 1820:
Heater extruder not heating at expected rate
Transition to shutdown state: Heater extruder not heating at expected rate
See the 'verify_heater' section in docs/Config_Reference.md
for the parameters that control this check.

[klippy.log](https://github.com/EtteGit/EnragedRabbitProject/files/7532518/klippy.log)

EtteGit, 2021-11-13:
Indeed, but the timer took close issue appears right at the call of the ERCF_HOME_EXTRUDER. I guess the issue comes from this function (located in the ercf.py). I don't know yet why it's happening to you

lerou114, 2021-11-22:
I found the issue was due to my own error in the extruder configuration my rotation distance was not set properly. I had upgraded all my firmware to a newer version prior to installing the ERCF. I did not do any prints without the ercf to verify my standalone configuration was properly working with the newest firmware. Once I revised the rotation distance this fixed the mcu time out issue.

closed 2021-11-22 by lerou114