EtteGit/EnragedRabbitProject #55: Screws for switchwire mounts exist... NO NUTS for said screws can be found

tnap1979, 2021-11-20:
i can locate the threaded printable screws for the ercf to mount to the frame, but cannot locate the nuts to lock them in place

EtteGit, 2021-11-20:
The threads are on the parts on the '2020' and '3030' folders (3030 being for the switchwire). There are no nuts.

tnap1979, 2021-11-20:
Oh jeez... wow.. I'm blind.. haha..
Ok, well in that case.. can close this..
Just built myself a 4port feeder and patch setup..
Dude... your design and setup of this device... blows my mind... are you an engineer of sorts?? Amazing work..  thanks for this project

EtteGit, 2021-11-20:
:) I'm glad you like it!

closed 2021-11-20 by EtteGit

tnap1979, 2021-11-20:
Forgot to ask.. as I'm unsure where to look...
Is there an afterburner motor mount plate that has a heatset insert for the toolhead board that supports the rabbit v3 toolhead board??
[image]

EtteGit, 2021-11-20:
You'll find what you want here I think https://github.com/VoronDesign/Voron-Hardware/tree/master/Afterburner_Toolhead_PCB/STLs/Clockwork

tnap1979, 2021-11-20:
I did.. thanks soooo much again..
😀