EtteGit/EnragedRabbitProject #60: Reducing change time

the-cobbler, 2021-12-01:
The ERCF is amazing, great work.
If you did not consider it, I may have an idea to speed up change:

 - Add a filament cutting mechanism
- Cut the current filament when the calculated material needed is in the length of the bowden (some custom code)
- Push new filament
- The new one will push the old one
- Purge the new one

This way the machine will keep printing while swapping, with a minimum time to optimize to spend on purging 

One downside is that if the swap need to happen every n < length of bowden you either will have to purge and waste filament or restrict the minimum amount of swap

Another downside is in flexibles, you will need to push the new one while the extruder pull the old one

EtteGit, 2021-12-01:
Thanks =)

The cut and use the filament until the extruder is something I've thought of quite some time ago, but the amount of work slicer // klipper wise to make it happen is THE issue I don't want to deal with right now.

Now if you, or anyone else, feels the courage to code that, I'll be happy to put a cuter in the ERCF 1.1 ^^

the-cobbler, 2021-12-01:
Give me a month or so and i'll be available :)
I have a current side quest of trying a POC of a bed probe using an accelerometer + gyroscope board

EtteGit, 2022-01-26:
Hi =) Just checking if I can close this issue or you still want it open.

Cheers

JiMcNulty, 2022-01-26:
Smash that close button

closed 2022-01-26 by JiMcNulty