EtteGit/EnragedRabbitProject #68: ERCF Overloaded?

Teila, 2022-01-27:
Greetings,

I have been getting an excessive amount of 'Timer too close - Shutting Down' issues when I am doing toolhead swaps during a print and when I am just manually cycling the tools for testing.

This is one of the latest logs after a crash.

[klippy (11).log](https://github.com/EtteGit/EnragedRabbitProject/files/7946596/klippy.11.log)
.

phil65, 2022-01-27:
Also experienced that issue once, in my case it was a loose cable connection of the infrared sensor. Perhaps worth checking.

Anathema-Device, 2022-01-27:
Hi Teila, I have seen this when trying to restart a test and also a divide by zero at the end of the calibration of feeder 0 when one would expect it to start feeder 1.  I updated to the latest ercf_software.cfg and ercf.py and that seems to have reduced the errors but ercf_calibrate does bizzare things now.  

It does the feeder 0 calibrate OK without the divide by 0 issue but then does some sort of partial calibrate on feeder 0 but the messages indicate the subsequent feeders, weird.

Teila, 2022-01-27:
> Also experienced that issue once, in my case it was a loose cable connection of the infrared sensor. Perhaps worth checking.

I have not seen any issue with my sensor disconnecting randomly. Perhaps a way to log the count rate vs the turn signals sent to the feeder on a load/unload for debugging is needed?

Teila, 2022-01-27:
> Hi Teila, I have seen this when trying to restart a test and also a divide by zero at the end of the calibration of feeder 0 when one would expect it to start feeder 1. I updated to the latest [...]

I also have had divide by zero on some calibration runs but whenever that happens I notice that is directly related to the filament not moving. I have had the servo not lock down when running the calibrate, the filament not get grabbed, filament get stuck temporarily in the extruder. However, those times it produces the divide by zero. My current issue occurs when I see the filament moving and the sensor (to my eye) is functioning (blinking) per normal.

EtteGit, 2022-03-17:
Hi,
I'm closing this thread since the other one is more active and concerns the same kind of issue (which should be fixed in the upcoming commits)

closed 2022-03-17 by EtteGit