EtteGit/EnragedRabbitProject #70: LGX filament sensor is not aligned with Afterburner/stealthburner

artheoday, 2022-02-04:
LGX mount in the filament sensor folder needs the spacer which is buried in the voron mods folder. Either include the spacer here or edit the STL so it wont need the spacer. 

This is the image of the user mod with ball and microswitch as compared to the official ERCF LGX mount 
[image]

EtteGit, 2022-02-22:
Indeed one will need the spacer, I'll add it to the repo.

RcTomcat1, 2022-05-17:
What spacer is needed? I already printed the official version and wanted to mount my LGX today. A search for "spacer" gives me only the bearing spacers. A short comparison between the user mod LGX mounts and the official still shows that height difference. Please advice on where to find that spacer

Benoitone, 2022-05-17:
You can find it in pin message in #voronuser_mods on discord...

RcTomcat1, 2022-05-17:
> You can find it in pin message in #voronuser_mods on discord...

Thanks. I will look there.
I searched through the enraged rabbit files + the voron files and no luck. 
The message above states that it was added to the repo so i assumed it would be in here somewhere.