EtteGit/EnragedRabbitProject #73: Newest Version of klipper breaking extras module

Beraval, 2022-02-11:
After Updating to the newest version of klipper I am presented with this error.

Internal error on command:"ERCF_HOME"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 245, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 415, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 186, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 213, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 186, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 213, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/ercf.py", line 394, in cmd_ERCF_SET_STEPS
    self.gear_stepper.rail.steppers[0].set_step_dist(new_step_dist)
AttributeError: MCU_stepper instance has no attribute 'set_step_dist'
Dumping stepper 'manual_stepper selector_stepper' (ercf) 2 queue_step:
queue_step 0: t=984515532 p=393 i=59994 c=-1187 a=0
queue_step 1: t=1007956721 p=3 i=0 c=0 a=0

phil65, 2022-02-11:
https://github.com/EtteGit/EnragedRabbitProject/pull/74

Teila, 2022-02-13:
Meaning that the ERCF modules still use the deprecated Set_Step_Dist. =[

I also am now getting an error with the new klipper update
[image]

Teila, 2022-02-13:
Attached Logs for those that would like it.
[moonraker.log](https://github.com/EtteGit/EnragedRabbitProject/files/8054812/moonraker.log)
[klippy (12).log](https://github.com/EtteGit/EnragedRabbitProject/files/8054813/klippy.12.log)

Anathema-Device, 2022-02-13:
https://github.com/EtteGit/EnragedRabbitProject/pull/74/commits/ff60a56bc3e6e2bf0fbb2385c55872d678c90078

EtteGit, 2022-02-22:
> 

The fix has already been merged on the ercf side, please update your ercf.py =)

EtteGit, 2022-02-22:
I'll keep this issue open for a few weeks, just for people to see easily what the fix is.

EtteGit, 2022-03-23:
Time to close this =)

closed 2022-03-23 by EtteGit