EtteGit/EnragedRabbitProject #82: ERCF_LOAD_FILAMENT_IN_EXTRUDER detect slow?

Teila, 2022-03-07:
Greetings!
Is there any way to add some buffer time to the detection of load into the nozzle? I find it causing a ercf_pause just before the filament detection sometimes.

Anathema-Device, 2022-03-07:
I have no solution but I see this too where an ercf_pause is invoked for some reason immediately followed by a filament loaded message.  My guess is that having the filament sensor on the hotend connected to my old 8-bit board is introducing some feedback lag as the load stepping continues on for a second or so AFTER the ercf_pause is issued.

Imnotkelby, 2022-03-29:
I am having the same problem and have tried adding pauses of lengths up to 2 seconds in between various parts of the ERCF_LOAD_FILAMENT_IN_EXTRUDER macro with no success, but maybe I missed something.

cmroche, 2022-05-07:
@EtteGit The same issue is happening with unload, and a fix (and explanation of the timing problem) is in PR #102 . I made the same fix for _ERCF_LOAD_FILAMENT_IN_EXTRUDER in that change too.

closed 2022-06-17 by EtteGit