EtteGit/EnragedRabbitProject #83: Servo not engaging sometimes

amv101, 2022-03-12:
I have been using the ERFC for a few weeks now, and have gotten most of the things worked out. I have an issue with the servo (MG90S) not engaging sometimes. It is wired directly into my Octopus v1.1 board 5V pin. It happens during feeding and unloading. I have tried all servo angles and top hat lockers, and have tried multiple cables with different lengths. Sometimes it just doesnt make the full swing for contact. The servo arm doesnt pass the bump. It will also do it randomly when homing ERCF or testing the grip. The only thing I havent changed was the servo.

Any ideas?

Benoitone, 2022-03-12:
Hello.
Have you read ERCF FAQ?

amv101, 2022-03-12:
> Hello. Have you read ERCF FAQ?

Looks like it has been updated since I have last checked. I see there might be a timing issue. I will check that after this current print is done. Thanks!

EtteGit, 2022-03-14:
> > Hello. Have you read ERCF FAQ?
> 
> Looks like it has been updated since I have last checked. I see there might be a timing issue. I will check that after this current print is done. Thanks!

Does the solution in the FAQ fixed your issue?

amv101, 2022-03-15:
> > > Hello. Have you read ERCF FAQ?
> > 
> > 
> > Looks like it has been updated since I have last checked. I see there might be a timing issue. I will check that after this current print is done. Thanks!
> 
> Did the solution in the FAQ fixed your issue?

I did a 6hr print without any issues. I had to turn the print off, and the printer is down for some maintenance. As soon as I get it back up I will test it some more this week.

I edited this line

```
    SET_SERVO SERVO=ercf_servo ANGLE={printer["gcode_macro ERCF_VAR"].servo_down_angle}
    G4 P200
```

Changed G4 P200 -> G4P205

EtteGit, 2022-03-23:
Everything is solved? Can I close this?

michaelbeljaars, 2022-05-31:
I have a similar problem. With top hats other then #1 the servo does not make the full swing. Changing angles in the configuration does not change that, nor does the proposed solution above of adjusting the timing. It seems the servo does not have enough power to fully press down the the top hat. I have found that top hat #1 does not give me enough pressure on most filaments, but switch to higher hats lead to even less pressure because of the issue with the servo.

When I manually position the servo arm to its full extension, I get the desired grip on the filament.
I am using a MG90S with an ERCF Easy BRD v1.1

Anathema-Device, 2022-05-31:
@michaelbeljaars I had a similar issue and I found that making sure that the top was smooth helped.  Adding a tiny bit of silicon grease or printing the lockers in resin made a big difference.  Also I found that it does not seem to take a huge amount of pressure to get a good grip but the arm has to be fully engaged so that it is in the correct place.

michaelbeljaars, 2022-05-31:
Thanks for your advice @Anathema-Device, I have tried much of what you suggest: smoothing the top hats (sanding, greasing) but that doesn't seem to change the fact that the servo arm can't pass the hump and subsequently relaxes back (and thus not applying pressure).

Anathema-Device, 2022-05-31:
@michaelbeljaars I think if it is catching in the dip in the top of the locker then the locker may be too high?  I also touched the top of each block and the bottom of the top hat with a file to make sure they were all smooth, flat and mated well.  If you lightly touch it the top hat should press down and engage easily but release when you remove your finger.  Everything has to be moving easily.  It is frustrating to get it all working.

michaelbeljaars, 2022-05-31:
No it's not catching in the top dip, it's just not able to push all the way to the lock position and hence it slides back, releasing pressure on the top hat. I've indeed noticed that filing the top hats makes a difference, which makes me question whether even the #1 top hat could be too high for my ercf.

Anathema-Device, 2022-05-31:
Make sure that the top and and the block below both have smooth, flat surfaces so that they mate cleanly and move smoothly.  You could also remove the top hat and make sure that the servo goes from the minimum position next to the servo body (servo_up) to full down position (servo_down) with no load as a start.  The servo down position will likely need to be a smidge further in order to work well.

dominikacri, 2022-06-01:
Had the same problem at the beginning. The servo pushed it perfectly to the lock position and then it was sliding a little bit back after that. The grip was not good. IIt was to weak with 1-3 and with 4 and aboth it was unable to reach the lock final position.
 
But I did many things wrong like tried to push the servo to 180 degrees. Tested the strength of the servo against my fingers and other bad things. So maybe I destroyed the servo with that :-)
 
I also ordered the mechanical parts as a set with the servo and maybe the MG90S was a fake. Not easy to know today.

So I reprinted the filament blocks, the lockers and the servo arm and this time the ERCF savox adapter with a better quality and ordered a savox servo which is also supported.

The MG90S has around 1.8kg/cm while the Savox SH0255-MG has around 3.1 kg/cm (depending which side you read more or less :-D)

I changed both together so I cannot say if the better print quality or the savox did the trick but after that the grip was very good.
 
I also ordered a L7806CV voltage regulator to replace the 5V that is used in the Easy BRD for powering the servo.
Both servos are capable of running with at least 6V with increased holding torque.

2.1kg/cm at 6V for MG90S  3.9kg/cm for the savox but that was not needed in my case. Only a minimal enhancement but dropping from 24 To 6V produces less heat than dropping from 24 to 5 so 2 good things I thought :-D
But I never used it.

But you can test it also with a step down. I saw that newer chinese boards uses also step downs instead of voltage regulators in their Easy BRD boards.

michaelbeljaars, 2022-06-01:
Here are two videos of the servo engaging on of the top hats. As you can seen in both videos it slides a bit back, although in one less than the other.

https://user-images.githubusercontent.com/28304614/171352892-e65e5fe2-8d1c-4012-8098-bc4a62a6360e.mp4

https://user-images.githubusercontent.com/28304614/171352896-aef513e0-8936-48ac-b9c8-e3805bf69395.mp4

I also discovered that some bontech gears were not perfectly aligned with the filament path in the blocks, so I fixed that, but it didn't seem to make much of a difference.

Without any top hat in front of it, the servo basically makes the full swing that it can make with the arm attached, meaning that it almost touches the selector in the down position.

https://user-images.githubusercontent.com/28304614/171353663-55899a61-a0a1-492b-b5a7-e122c27ce2c5.mp4

michaelbeljaars, 2022-06-07:
I just replaced my MG90S for a Savox SH0255-MG, but that made no difference. The servo is still unable to push the top hats fully to the lock position and then slides back.

michaelbeljaars, 2022-06-08:
I have adjusted the hats by creating a more pronounced slot at where the servo arm should rest in the lock position using a circular file. This seems to have solved the problem with the servo arm not getting to the lock position and also results in enough pressure on the filament.

EtteGit, 2022-06-17:
> Here are two videos of the servo engaging on of the top hats. As you can seen in both videos it slides a bit back, although in one less than the other.
> 
>  VID_20220601_093243.mp4 
>  VID_20220601_093439.mp4 
> I also discovered that some bontech gears were not perfectly aligned with the filament path in the blocks, so I fixed that, but it didn't seem to make much of a difference.
> 
> Without any top hat in front of it, the servo basically makes the full swing that it can make with the arm attached, meaning that it almost touches the selector in the down position.
> 
>  VID_20220601_094331.mp4 
> I have also found that without any filament inserted, the servo arm is more likely to get into the correct down position (although not always). When subsequently inserting filament by manually turning the bontech gears, it can be seen that the selector moves somewhat outward as the filament passes between the gears.

Hi! Your down angle is too "high", i.e. it goes too far. You want the down angle to set the servo arm in a "vertical" position, kind of like where the servo arm ends up after slipping in your videos. This is at this position that the grip will be the best. Also from this position it shouldn't slip anymore

EtteGit, 2022-06-17:
> I have adjusted the hats by creating a more pronounced slot at where the servo arm should rest in the lock position using a circular file. This seems to have solved the problem with the servo arm not getting to the lock position and also results in enough pressure on the filament.

Just to mention that, quite some time ago, top hats lockers had such "locking slots" on them. Turned out the bump required to make such a slot was too much of an effort for the servo to overcome for many people, hence this design choice was abandoned.

amv101, 2022-08-01:
> Hi! Your down angle is too "high", i.e. it goes too far. You want the down angle to set the servo arm in a "vertical" position, kind of like where the servo arm ends up after slipping in your [...]

@EtteGit  I am still having some issues with the servo. Sometimes there is an issue as described above, and sometimes the servo doesnt make the full sweep at all. As for the slippage, it seems the servo is disengaged after the servo arm move which causes the servo to slowly move out of position due to vibrations. Is there a way to keep the servo powered why it is engaged? I have built many projects with servos and usually the servo stays powered after a move to make sure the position doesnt change.

amv101, 2022-08-07:
> Here are two videos of the servo engaging on of the top hats. As you can seen in both videos it slides a bit back, although in one less than the other.
>
>  VID_20220601_093243.mp4 [...]

@michaelbeljaars Any luck finding out why the servo moves out of position, like a twitch, sometimes after the servo movement? I have 3 ERCF units about 5 months old, and they have always done this behavior. It is not consistent though. Sometimes I can get through a full print without any errors. Sometimes the error plaques me all day.

michaelbeljaars, 2022-08-28:
@amv101 No, not yet. I've had a few weeks of summer holidays (I work at a vocational school). What I've done last is file a little dip in the top hats where the servo arm can hook behind. It is not consistent though and as @EtteGit mentioned, sometimes the locking force is too great for the servo to get to the unlock position.

It seems that for my setup already the #1 top hats are too high for the servo to push them fully into position. I am not sure yet what I will be trying next.

dominikacri, 2022-10-11:
Re: [EtteGit/EnragedRabbitProject] Servo not engaging sometimes (Issue #83)

Had the same problem at the beginning. The servo pushed it perfectly to the lock position and then it was sliding a little bit back after that. The grip was not good. IIt was to weak with 1-3 and with 4 and aboth it was unable to reach the lock final position.
 
But I did many things wrong like tried to push the servo to 180 degrees. Tested the strength of the servo against my fingers and other bad things. So maybe I destroyed the servo with that :-)
 
I also ordered the mechanical parts as a set with the servo and maybe the MG90S was a fake. Not easy to know today.

So I reprinted the filament blocks, the lockers and the servo arm and this time the ERCF savox adapter with a better quality and ordered a savox servo which is also supported.

The MG90S has around 1.8kg/cm while the Savox SH0255-MG has around 3.1 kg/cm (depending which side you read more or less :-D)

I changed both together so I cannot say if the better print quality or the savox did the trick but after that the grip was very good.
 

I also ordered a L7806CV voltage regulator to replace the 5V that is used in the Easy BRD for powering the servo.
Both servos are capable of running with at least 6V with increased holding torque.

2.1kg/cm at 6V for MG90S  3.9kg/cm for the savox but that was not needed in my case. Only a minimal enhancement but dropping from 24 To 6V produces less heat than dropping from 24 to 5 so 2 good things I thought :-D
But I never used it.

But you can test it also with a step down. I saw that newer chinese boards uses also step downs instead of voltage regulators in their Easy BRD boards.

Make sure that the top and and the block below both have smooth, flat surfaces so that they mate cleanly and move smoothly. You could also remove the top hat and make sure that the servo goes from the minimum position next to the servo body (servo_up) to full down position (servo_down) with no load as a start. The servo down position will likely need to be a smidge further in order to work well.
—

Mit freundlichen Grüssen
Dominik Acri