EtteGit/EnragedRabbitProject #84: Pause commend causing "Move out of range:" issue

amv101, 2022-03-12:
I am running ercf on a custom Delta printer. When a clog is detected, or there is no filament in the cart, I will get a "Move out of range" error when the printer goes into the pause state. I have tried reducing my delta radius by half, to see if there was an issue with the probe offset, but I still get the same error.

Any ideas what the issue might be?

Error from the Octoprint terminal when sending "pause" right after homing / moving down

```
Recv: // Klipper state: Ready
[...]
Send: G28 // homes all axis
Recv: ok
Send: G91 // set absolute move
Recv: ok
Send: G0 Z-50 F1000 // move down 50mm
Recv: ok
Send: G90 // set relative move
Recv: ok
Send: G91
Recv: ok
Send: G0 Z-100 F1000
Recv: ok
Send: G90
Recv: ok
[...]
Send: pause
Recv: // action:paused
Printer signalled that it paused, switching state...
Recv: !! Move out of range: -145.000 145.000 245.085 [-1.000]
Recv: !! Move out of range: -145.000 145.000 245.085 [-1.000]
Recv: ok
```

printer.cfg

```
# This file contains common pin mappings for the BigTreeTech Octopus.
# To use this config, the firmware should be compiled for the
# STM32F446 with a "32KiB bootloader" and a "12MHz crystal" clock reference.

# See docs/Config_Reference.md for a description of parameters.

[include ./ercf_hardware.cfg] # ERCF settings
[include ./ercf_software.cfg] # ERCF settings

# Driver0
[stepper_a] # StepperOnline 17HS19-2004S1, 1.8 degree, 2A
step_pin: PF13
dir_pin: !PF12
enable_pin: !PF14
microsteps: 16
rotation_distance: 40
endstop_pin: PG6
#position_endstop: 385
homing_speed: 80
homing_retract_dist: 5.0
second_homing_speed: 3
#arm_length: 400

# Driver1
[stepper_b] # StepperOnline 17HS19-2004S1, 1.8 degree, 2A
step_pin: PG0
dir_pin: !PG1
enable_pin: !PF15
microsteps: 16
rotation_distance: 40
endstop_pin: PG9

# Driver2
[stepper_c] # StepperOnline 17HS19-2004S1, 1.8 degree, 2A
step_pin: PF11
[... 490 lines cut ...]
#*# distance4_pos1 = 31437.324,31331.721,30877.190
#*# distance4_pos2 = 36452.654,36348.423,27444.518
#*# distance5 = 116.89
#*# distance5_pos1 = 31057.212,31521.233,31064.766
#*# distance5_pos2 = 30118.752,40238.954,30133.731
#*# distance6 = 117.08
#*# distance6_pos1 = 27642.962,35225.118,36049.904
#*# distance6_pos2 = 30390.832,29717.629,39942.649
#*# distance7 = 116.9
#*# distance7_pos1 = 30363.754,29692.952,39096.539
#*# distance7_pos2 = 36565.297,27410.911,35855.203
#*# distance8 = 117.04
#*# distance8_pos1 = 36066.550,27553.877,35373.634
#*# distance8_pos2 = 39974.274,30306.112,29848.646
#*# distance9 = 116.73
#*# distance9_pos1 = 39122.926,30278.013,29822.474
#*# distance9_pos2 = 35859.192,36446.060,27515.873
#*# distance10 = 116.95
#*# distance10_pos1 = 35376.072,35949.256,27659.066
#*# distance10_pos2 = 29833.240,39801.402,30394.945
#*# distance11 = 116.9
#*# distance11_pos1 = 29807.322,38958.887,30368.277
#*# distance11_pos2 = 27499.425,35702.979,36545.839

```

FHOGPHIL, 2022-03-12:
Are you using the pause and resume macros specifically for ERCF?

RomRider, 2022-03-13:
It's probably related to being in relative vs absolute positioning somewhere in your macros

amv101, 2022-03-13:
There is definitely something up with that setting. From some searching, it has something to do with the probe offsets and the mesh possibly. My Delta has a print radius of 150mm and a print height about 340mm. I run the delta calibration and mesh at 130mm.

When I use this setting, the issue goes away…

```
[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
gcode:
  {% set x_park = printer.toolhead.axis_minimum.x|float + 55.0 %}
  {% set y_park = printer.toolhead.axis_maximum.y|float - 55.0 %}
```

Now it will park at -95.00 95.00 without any issues.

Maybe I should run the delta calibrate and bed mesh with the same radius as the print radius?

amv101, 2022-03-13:
> Are you using the pause and resume macros specifically for ERCF?

Yes... I run the "pause" command from the Octoprint terminal.

FHOGPHIL, 2022-03-13:
> > Are you using the pause and resume macros specifically for ERCF?
> 
> Yes... I run the "pause" command from the Octoprint terminal.

That’s not what I asked. Did you replace the klipper pause ad resume macros with the ones from the ERCF GitHub?

amv101, 2022-03-13:
> > > Are you using the pause and resume macros specifically for ERCF?
> > 
> > 
> > Yes... I run the "pause" command from the Octoprint terminal.
> 
> That’s not what I asked. Did you replace the klipper pause ad resume macros with the ones from the ERCF GitHub?

I assume so, but I am pretty new to Klipper. I never added any pause macros to my printer.cfg file, except for the ercf macros. I added the ercf macros and then type "pause" into the terminal via Octoprint.

My cfg file is posted above with the macros I have. Would there be another place to check for conflicting klipper pause macros?

amv101, 2022-03-15:
> The "ercf compatible" pause and resume macros are in the "client_macros.cfg" file on the ercf repo (klipper file folder).
> 
> Those are the one you need to use with the ercf (or adapt the ones you have to contain the ercf stuff).

Thanks for the help. I posted my cfg above. Did you see an issue with how I added the ercf macros? Not sure why I am getting the error on my delta. Does it have something to do with my bed mesh radius, delta calib radius, and delta print radius being different?