EtteGit/EnragedRabbitProject #85: Ptfe tube from hotend cant enter the cw1

Z3ytrox, 2022-03-14:
The ptfe tube of the hotend cant enter the ercf ab cw1 extruder. Therefor Its pressed against the hole that sometimes lead to failed loads. The filament is sometimes pushed against the ptfe tube instead of inside the ptfe tube. Any fixes for this?

Z3ytrox, 2022-03-14:
[image] (Normal AB)
[image]
(ERCF AB)

claytono, 2022-03-14:
This is called out on page 76 of the manual.  You have to use a shorter PTFE tube.

Z3ytrox, 2022-03-14:
Yeah ive seen that but how do you solve the issue thats caused by this? The filament thats running into the ptfe instead of through it?

closed 2022-03-17 by EtteGit