EtteGit/EnragedRabbitProject #87: ERCP: Buffer_Wheel does not "fit" into Buffer_Cross cavity

fabikroeger, 2022-03-16:
Hi There, 

i have been assembling the ERCP. While doing so, i have experienced, that the buffer_wheel does not spin freely as i would like to have it spinning. I spotted a friction between Buffer_Wheel and Buffer_Cross. With a closer look it seems for me that there is a cavity in which the buffer_wheel shoult spin but it is not big enought. As marked on the screenshot. 
[image]

[image]

For my understanding the fix would be to have a cylinder, slightly bigger than the buffer_wheel substracted from the buffer_cross.
I did that and it looks good for me for now but i want to ask if i do not understand my right.
The next picture shows a fix, which  would be to have a cylinder, slightly bigger than the buffer_wheel substracted from the buffer_cross.
I did that and it looks good for me. 
Pink: old buffer_cross
Grey: new buffer_cross
[image]

Do we have the slightly smaller cavity for the buffer_wheel by intention or is my fix doing his job?

Greetings

FHOGPHIL, 2022-03-16:
[image]
[image]
All nine of these move fine. Any chance you over tightened the center bolt?

fabikroeger, 2022-03-16:
Ah ok thanks for the fast answear, less surface == less friction. Thx! hm, may be yes. 
I am building this here : https://github.com/probably-Erwins-Cat/ERCF-Buffer-Array
I need a buffer on the outside of my filament box.  
And maybe i have by its design more friction there.
I will have a check on that tomorrow and come back then.

TravisWilder, 2022-03-16:
> Any chance you over tightened the center bolt?

but also let it to loose as else the filament can make it beside the buffer wheel

fabikroeger, 2022-03-17:
Okay, fixed. Thank you all for the hints and suggestions!! 
1. To much infill for my first layer of the buffer wheel. So it has had to much friction. The knife helped there.
2. The centerbolt is now proper tightened.

fabikroeger, 2022-03-18:
-