EtteGit/EnragedRabbitProject #95: pin_override

Dragi2k, 2022-04-08:
i've noticed a problem that is possible in the klipper config.

The pin_override can only be onced in a config, the second one will be ignored.

On the Documentation should be a part that if somebody noticed that klipper says error pin pa6.. that you should look into your config that pin override is probably in use and the second will be ignored.

Anathema-Device, 2022-04-08:
There can only be one, the last one read as the config is processed is the one used.  Put all your overrides into one statement.

Dragi2k, 2022-04-09:
Yea, now i know That too 😂

EtteGit, 2022-06-17:
Hi, you need to put all the pins you want to duplicate into the same [duplicate_pin_override]. Iirc just list them with coma in-between ;)

closed 2022-06-17 by EtteGit