EtteGit/EnragedRabbitProject #96: Internal error during connect: MCU_stepper instance has no attribute 'steppers

PatrickYCY, 2022-04-14:
When include ercf's config have error as the tittle,
Attached the Klippy log for reference
Machine: Voron 2.4 350, MCU 1 BTT OCTOPUS V1.1, MCU 2 BTT Mini SKR E3 V2.0
[klippy (2).log](https://github.com/EtteGit/EnragedRabbitProject/files/8490604/klippy.2.log)

EtteGit, 2022-04-14:
Please include the ercf cfg files at the end of your printer.cfg, not at the beginning

PatrickYCY, 2022-04-15:
> Please include the ercf cfg files at the end of your printer.cfg, not at the beginning

I tried include the ercf cfg files at the end of the printer .cfg ,but with same error 
attached new klipper log for reference
[klippy (3).log](https://github.com/EtteGit/EnragedRabbitProject/files/8493617/klippy.3.log)

Benoitone, 2022-04-15:
Hello, if you name your mcu ercf you must adapt the pin with ercf to allow klipper to know on which mcu your ERCF is connected to.

PatrickYCY, 2022-04-15:
> 

i confirm that the ercf pins and config is declare correctly the mcu sections, and still error

cmroche, 2022-04-27:
This will be fixed by #100