EtteGit/EnragedRabbitProject #107: MCU 'mcu' shutdown: Timer too close when select T1

PatrickYCY, 2022-05-17:
When i select the T1 slot th machine will show MCU 'mcu' shutdown: Timer too close 
Please find the attached file for you action

[klippy (1).log](https://github.com/EtteGit/EnragedRabbitProject/files/8710639/klippy.1.log)

Anathema-Device, 2022-05-17:
Have you looked at:
https://github.com/EtteGit/EnragedRabbitProject/issues/75 or
https://github.com/EtteGit/EnragedRabbitProject/issues/68

cmroche, 2022-05-26:
Based on this line in the log, it looks like the timeout happens after starting a long move. It's the last command sent before the timeout I believe:

```
Sent 99 176.661427 176.658867 64: seq: 1e, queue_step oid=9 interval=64009049 count=1 add=0, queue_step oid=9 interval=8821 count=11 add=-617, queue_step oid=9 interval=3462 count=23 add=-75, queue_step oid=9 interval=2028 count=42 add=-18, queue_step oid=9 interval=1366 count=69 add=-6, queue_step oid=9 interval=1009 count=76 add=-3, queue_step oid=9 interval=826 count=73 add=-2, queue_step oid=9 interval=704 count=130 add=-1, queue_step oid=9 interval=599 count=74 add=-1
```

This would indicate that you are hitting issue #75

EtteGit, 2022-06-17:
I'm closing this thanks to the nice @cmroche PR for the long move splitting trick =)

closed 2022-06-17 by EtteGit