EtteGit/EnragedRabbitProject #110: ERCP High Friction

Durahl, 2022-05-26:
Greetings!

Been using the ERCF / ERCP for I think like 9 months or so and the StealthBurner / ClockWorkd2 on my Voron 2.4 since it was released to the Public with my own MODs to make the CW2 ERCF compatible.

Two exhaustive printing weeks must have worn down my Beta 0/1 CW2 Toolhead Parts down to the point where - I assume - The Guidler and/or Latch may have gotten bent out of shape just enough for the Gears to not be digging in enough anymore to overcome the always present friction in the ERCP > ERCF > CW2 Filament Path thus resulting in frequent Filament Grindings and failed Prints - With the necessity of printing new parts I decided to give the recent CW2 Beta 7-11 Parts a try which turned out to be even worse but while I was it I went ahead and traced the cause of the friction back to the off tangential Filament Exit Port of the ERCP:
[image]
Basically... Whenever the Filament was stuck and ground to an Hourglass shape in the Toolhead I checked where it may have been stuck by pushing it back but couldn't feel any resistance but pulling it beyond the point where it became stuck was almost impossible.
Taking the Buffer Zone out of the equation would mostly solve this so the issue must be there ( my wheels are spinning freely and the Spools too have been placed on bearings.
For the time being I've resorted to not be winding the Filament onto the wheel and instead just use two PTFE Tubes with one connected to the ERCP PTFE Entry Block and the other to the ERCF Filament Block causing them to split apart if the Filament is being retracted and butt together when the FIlamen is being Extruded.

I'll probably do a DIY MOD of the ERCP to see if this can be solved by printing a Buffer Cross with the Entrance / Exit ports pointing perpendicular towards the Wheels Inner Diameter and thus not introducing that kink in the Screenshot.

I've also noticed certain Filament having a tendency of liking to get pushed out between the rear and bottom Cross sections thus spilling outside instead of coiling up inside - Maybe a wall connecting/closing those two could help with this? 🤔

Anathema-Device, 2022-05-26:
I have a similar issue and tried simple mods to let me push the PTFE further through with the hope that it would result in less friction but that was not 100% successful either.  What I observe that in the jammin' case it seems that coils have contracted overlapping so that there is much increased drag and it stalls the extrusion.    Would a slight thread on the wheel help to set up the path better?  Maybe that with an adjustment like you describe would make it work more reliably.

Personally though I'm looking at the enclosed vertical columns that some have demo'd as a way to contain the retracted filament and keep it isolated and dry.

Durahl, 2022-06-03:
So, here's my take on the suggested changes...

In the first two pictures the Filament now enters the PTFE Tube without the chance of forming a Kink:
[image]
[image]

The third picture shows the restraining "fence" I was talking about:
[image]
May have to add a ridge to it if the Filament slides off which is like a sub 5min print...

Not pictured but I've also changed the entry Geometry of the PTFE Slider on the Arm to the shape of a Funnel gradually feeding into the PTFE Tube after noticing the orifice on the stock ones having developed a deep groove from the filament sliding through it potentially being the trigger for the kink later down the PTFE road 😑

Currently testing how it is going 😁

I'll probably have to have a look into the ERCF itself if there's a similar problem with the filament having carved a groove at some location creating another Friction Hotspot 🤔

EtteGit, 2022-06-17:
Nice work on this ! Don't forget to do a PR for this Carrot Patch mod, I want to try those changes =)

jwhite, 2022-08-01:
I too would like a PR.  I'm finding this is my #1 source of failure, and am looking at the geometry in fusion.

[image]

NameOfTheDragon, 2022-09-21:
Interesting. I have had similar problems with high friction, but I have traced it to a different source: the wheel. On the inside of the cross-arm, there is a circular inset that is a little smaller then the wheel. The rim of the wheel rubs up against the cross arm generating friction, and this will be particularly bad if you printed on textured PEI. This has always seemed like an odd design choice to me, as it tends to undermine the inclusion of a ball bearing on the axle. Tension in the filament at this point increases friction all the was along the rest of the feed path, so the closer to the spool it can be eliminated, the better. My usermod is submitted at PR #155 if you'd like to give it a go. Since we have edited the same part, it might be worth combining these mods at some point, once they have got some testing and feedback from users.

jwhite, 2022-09-22:
Thanks @NameOfTheDragon - I was noticing that I have two points of friction, one where the coils have yet to align, and then later once they have all gotten tight.  I wonder if this might resolve the latter?  It's very strange because sometimes there is almost no friction when manually pulling the filament from the extruder end of the tube and sometimes it won't even budge.  I'm using prints on a textured PEI - so I would be seeing what you might have resolved.  Let me print off one of yours combined with one of mine.  

Cheers!

3D-Printing-Nerds, 2022-12-30:
Durahl Can we get a copy of the files for your mod on the clockwork2 
I want to build a stealthburner to use with my ERCF but i cant find one with the Hall Sensor built in to it