EtteGit/EnragedRabbitProject #111: ECAS vs. PC4-M10 on Carrot_Patch/STLs/Buffer_Cross.stl

Jaans, 2022-05-28:
Hello Ette - more of a question than an issue but thought the response might help others.

Between all the different fittings (ECAS, Pressfit, UM2, PC4-M10, PC4-M6, Straight-through, etc.) I'm trying to understand if there is a particular benefit of using ECAS over say PC4-M10?

I happen to have a number of PC4-M10 fittings on-hand and wondered if I could adapt the step files (my CAD skills permitting) and change the Buffer Cross to use a PC4-M10 fitting instead of ECAS?

Could you share some insight of why ECAS might be better? Also, when is the straight through connecter better (or even why does it exist)?

PS: I'm going down the proverbial "rabbit hole" 🐰 with ERCF (waiting on parts) & ERCP (using Ikea 365+ 10l drybox) and enjoying it very much. Thank you (and all that helped) for creating and sharing something amazing!

Anathema-Device, 2022-05-28:
Having PC4-M10 fittings an no extra ECAS I just tapped an M10 thread into the ECAS hole and screwed it in.  Seems to be holding ...

Jaans, 2022-05-30:
Ooo... never thought of that. No need to mod the CAD designs then. I'll give that a go! Thank you 👍 

I see no obvious functional difference between ECAS04 and a PC4 type fitting in that the one doesn't have better "holding" capability than the other. In fact, the PC4 type seems more convenient as it doesn't require a clip because it is spring loaded.

All I can think of is that ECAS04 is cheaper, lighter, smaller - which makes sense for it's use on the Carrot Feeder.

closed 2022-06-05 by Jaans