EtteGit/EnragedRabbitProject #115: Num moves

RomRider, 2022-06-20:
I think #113 is missing so parts to be fully effective. `MOVES=...` is not applied to all ERCF_LOAD/UNLOAD calls

What about overloading the ERCF_LOAD and ERCF_UNLOAD macro instead ?

Like so:
```
[gcode_macro ERCF_LOAD]
rename_existing: _ERCF_LOAD_ORIG
gcode:
  _ERCF_LOAD_ORIG {rawparams} MOVES={printer["gcode_macro _ERCF_VAR"].num_moves}
```

Same for ERCF_UNLOAD ?

Should I open a PR?

EtteGit, 2022-06-23:
I think the idea was just to hotfix the problematic calls. But I don't mind adding it to all ERCF_LOAD // UNLOAD calls if that causes no issue.
You can open a PR and test things indeed =)

RomRider, 2022-06-23:
The thing is that the PR doesn't fix all the long moves (eg.: the unload ones).
I'll put that in the python code directly it's more clean.