EtteGit/EnragedRabbitProject #120: Implement slaved extruder using gear motor to improve filament feed performance

baccula, 2022-07-10:
High friction filament paths (particularly when using ERCP) plague many ECRF setups. Klipper offers a feature which allows [additional stepper](https://www.klipper3d.org/Config_Reference.html?h=extr#extruder1) motors to be slaved to the extruder during printing. 

Implementing this method using the gear selector motor to provide additional filament pushing force would mitigate filament drag issues, however it could also introduce issues with filament bunching or being stretched if the stepper calibration between the two motors is not 100% accurate. To overcome this, I suggest adding a periodic 'SERVO UP' command to remove tension and allow the filament path to balance. This could be done iteratively and in theory should not have any affect on the print GCode.

As not all users will need or benefit from this type of setup (or be able to achieve the close calibration tolerances necessary), I suggest using a configurable parameter in ecrf_software.cfg to enable or disable this behavior. In addition, a configuration field allowing user-specified intervals between servo cycles may be beneficial (alternatively, it may be better to just call a servo cycle for every 'x' mm3 of extruded filament, if such a thing is possible).

EtteGit, 2022-07-17:
Hi,

As discussed in the discord channel, I think this is totally doable. The only issue this can cause is during a clog. I'm afraid of filament grinding by the ERCF gears before the encoder can sense the clog. But this is something worth a try! Have you tried something alredy?

RickDnamps, 2022-11-14:
Any news for this Enhancement ??? I would be a nice add-on !

RickDnamps, 2022-11-15:
I tried quickly to do So but we cannot declare a pin two times in the .CFG :( I wanted to do it quickly by simply make it follow the principale extruder (there is a new klipper code who allow it now) and activate the servo and release it sometime to remove any tension or synchro. I'm not the best in programmation maybe I should work harder on that!

baccula, 2022-11-15:
Klipper supports duplicate pin configuration (see here
<https://www.klipper3d.org/Config_Reference.html#duplicate_pin_override> for
info). Not sure if that would accomplish what you're looking for or not.

RickDnamps, 2022-11-16:
I just try the dupplicate pins like that:

[duplicate_pin_override]
pins: ercf:gpio10, ercf:gpio9, ercf:gpio8, ercf:gpio20

but that doesn't work.

Still have error message:
!pin gpio10 used multiple times in config

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted!

I try to use the gear stepper motor of the ERCF to help the filament by pushing it and by synchronize it with the main extruder.... if you can help me !!

porklander, 2022-12-28:
> I just try the dupplicate pins like that:
> 
> [duplicate_pin_override] pins: ercf:gpio10, ercf:gpio9, ercf:gpio8, ercf:gpio20
> 
> also tried it without ercf in front...
> 
> but that doesn't work.
> 
> Still have error message: !pin gpio10 used multiple times in config
> 
> Once the underlying issue is corrected, use the "RESTART" command to reload the config and restart the host software. Printer is halted!
> 
> I try to use the gear stepper motor of the ERCF to help the filament by pushing it and by synchronize it with the main extruder.... if you can help me !!

Do you have success with this?

Anathema-Device, 2022-12-28:
Do you have more than one [duplicate_pin_override] defined maybe?