EtteGit/EnragedRabbitProject #121: Use ERCF as bowden extruder?

xeroe, 2022-07-11:
Heyho,
Id like to use the design in my bowden printer.
Do you think it will be possible to use the ercf as the bowden extruder or will i have to reinforce the gear/servo assembly to have more pressure?

Anathema-Device, 2022-07-11:
I don't see why it would not work.  You would need to think of a way to put the filament sensor near the hotend I expect.  Hypothetically the tube lengths may even be shorter requiring less turns on the takeup thus less drag than other arrangements?

EtteGit, 2022-07-17:
It's most likely possible, with some minor software changes. Though, concerning your pressure question, my guess is that this will heavily depends on the speed you want to print at. The higher, the more pressure// grip you'll need.

StanislavJochman, 2023-08-13:
@xeroe Hi. In theory it should be possible, but in practise I would discourage that. There are multipe possible problems. 

- Servo would need to be engaged whole time while printing which can lead to it burning out. 
- ERCF uses bondech gears for extrusion which is fine for just starting the filament but better solutions exsist for extruders. It would not be hight performance
- Biggest problem which I see is in bouden setup, because multi material printing requires a lot of precesion to be done well and using bouden limits that. It can also cause stringing color bleed and other artefacts.

xeroe, 2023-08-13:
didnt expect an answer after a year but yes i agree with you and ditched the idea :D

StanislavJochman, 2023-08-13:
Please close the issue.