EtteGit/EnragedRabbitProject #122: with mm print if i resume after pause with ercf I get error can't move out of range. With single printing pausing works great.

rwijnhov, 2022-07-14:
with mm print if i resume after pause with ercf I get error can't move out of range. With single printing pausing works great. I use klipper on a ratig 400.

closed 2022-07-14 by rwijnhov

heschlie, 2022-11-08:
@rwijnhov Can you share what you did to resolve this, I'm currently having the exact same issue and am running out of ideas. I assumed it was in relative mode instead of absolute but I tossed in a `G90` in the resume macro and am still seeing this. Thanks!

heschlie, 2022-11-08:
Would you be willing to share your pause/resume macros? I assume that is where my problem is but I am having trouble sorting it out.

Anathema-Device, 2022-11-08:
Make sure that your slicer is using relative extrusion rather than absolute?

rwijnhov, 2022-11-09:
hi i use this on my ratrig 400
############################
####ERCF Cancel Print Resume
#############################
[gcode_macro CANCEL_PRINT]
description: Cancels the printer
rename_existing: CANCEL_PRINT_BASE
gcode:
  END_PRINT
  TURN_OFF_HEATERS
  CLEAR_PAUSE
  #SDCARD_RESET_FILE
  SET_FILAMENT_SENSOR SENSOR=encoder_sensor ENABLE=0
  CANCEL_PRINT_BASE

[gcode_macro PAUSE]
description: Pauses the printer
rename_existing: PAUSE_BASE
variable_extrude: 1.5
gcode:
  SAVE_GCODE_STATE NAME=PAUSE_state
  # Define park positions 
  {% set E = printer["gcode_macro PAUSE"].extrude|float %}
  {% set speed = printer["gcode_macro RatOS"].macro_travel_speed|float * 60 %}
  # Calculate safe Z position
  {% set max_z = printer.toolhead.axis_maximum.z|float %}
  {% set act_z = printer.toolhead.position.z|float %}
  {% if act_z < (max_z - 2.0) %}
      {% set z_safe = 2.0 %}
  {% else %}
      {% set z_safe = max_z - act_z %}
  {% endif %}
  PAUSE_BASE
  G91
  # Retract
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G1 E-{E} F2100
  {% else %}
    {action_respond_info("Extruder not hot enough")}
  {% endif %}
  # Move to park position
  G90
  {% if "xyz" in printer.toolhead.homed_axes %}
    G1 Z{z_safe} F900
    _PARK LOCATION={printer["gcode_macro RatOS"].pause_print_park_in} X={printer["gcode_macro RatOS"].pause_print_park_x}
  {% else %}
    {action_respond_info("Printer not homed")}
  {% endif %} 

[gcode_macro RESUME]
rename_existing: BASE_RESUME
gcode:
    {% if printer.ercf.is_paused|int == 1 %}
        M118 You can't resume the print without unlocking the ERCF first.
        M118 Run ERCF_UNLOCK and solve any issue before hitting Resume again
    {% else %}
        RESTORE_GCODE_STATE NAME=PAUSE_state
        G90
        {% if printer.ercf.clog_detection|int == 1 %}
            SET_FILAMENT_SENSOR SENSOR=encoder_sensor ENABLE=1
        {% endif %}
        BASE_RESUME
    {% endif %}

heschlie, 2022-11-09:
Thanks @rwijnhov I also decided to try out the sensorless branch (didn't realize that was a thing) and that seems to have fixed it for me. I'm glad to see another ratrig user adding an ERCF to their machine!