EtteGit/EnragedRabbitProject #131: Acceleration when synchronize toolhead and manual stepper

eamars, 2022-07-31:
[image]
When toolhead and gear stepper is asked to move synchronously, the short move acceleration is specified to the gearstepper while the toolhead keeps its own acceleration. I suspect this could cause "Timer too close" issue when the microstep is setting beyond 16. 

In this case my solution is to set the same acceleration that toolhead uses to the gearstepper. This fix allows me to set more than 16 microsteps when initializating the gear stepper. 

Do you guys have any comment here?

EtteGit, 2022-08-04:
We could indeed match both motors accels, that looks like the most logical option.

Though the accel difference is not supposed to cause any issues. The "Timer too close" on high microsteping is just the result of the high microsteping itself : it takes longer for the system to generate the steps, and they are sent a tiny bit too late, hence the error.

Same issue arises if you increase the microsteps on the x or y stepper for instance.

eamars, 2022-08-04:
Thanks for the heads up. I tends to agree with your conclusion that the microcontrollers can't keep up. However I do observe the timer too close issue can be minimised when two steppers are sharing the same acceleration. 

I'll continue monitoring this issue on my branch of the ERCF driver.