EtteGit/EnragedRabbitProject #137: Asking for the official clockwork2 version filament sensor STL

PatrickYCY, 2022-08-14:
The stealthburner is already release the RC version for a few days ago ,so may i ask  when will the EnragedRabbitProject project team will release the clockwork 2 version filament sensor stl?

ccfman2004, 2023-02-12:
I've been wonder this too.

Adri-Donn, 2023-03-29:
+1 :D

lisander-lopez, 2023-04-14:
+1 <3

Anathema-Device, 2023-04-14:
I wonder if it is actually necessary.  I don't run a CW2 right now but after moving to the sensor-less version the ERCF has never been more reliable.  So I really don't miss the tool-head filament sensor.  I was running with the CW1 with the ball bearing and microswitch version of the sensor.

AustinTylerDean, 2023-06-28:
I would love an update on this as well. I've been hunting around and considering it's 2023 and I'm having trouble finding a bunch of people solving this "problem", it makes me think that most of those willing to go this far DOWN THE RABBIT HOLE haven't found this to be an issue. I've got some questions up on the discord server but for now, I would love an official weigh-in from the author @EtteGit !

Anathema-Device, 2023-06-28:
It really is not needed, moving to Happy Hare V3 code to drive the ERCF makes the tool head sensor unnecessary
https://github.com/moggieuk/ERCF-Software-V3