EtteGit/EnragedRabbitProject #141: Skipping the Daughter Board?

Durahl, 2022-08-24:
Is it possible - with a Gear Ratio MOD to the ERCF - to run both the ERCFs Gear _**Motor**_ and the Tool Heads Extruder _**Motor**_ in Sync with each other and thus off the same Stepper _**Driver**_?

I'm currently using a **BTT _Octopus V1.1_** on a **_Voron 2.4_** together with an _**ERCF Easy Board**_ but the latter is giving me grief as in the past two weeks or so every once in a while the system is loosing its connection to it which brings me to the idea of using the Octopus for the ERCF as well after all but with it only having 8 Stepper _**Drivers**_ ( _but_ 9 _**Connectors**_ ) and the Voron 2.4 already using 7 of those _**Drivers**_ we're technically 1 short unless we're using the 2 Connectors that share the same Driver for both the ERCF Gear Motor and the Extruder Motor?

Would this be a viable solution? 🤔
My thinking is that the ERCF Servo would disengage from the Filament anyway after the latter was loaded so even if for whatever reason the two BMGs wouldn't be perfectly in sync it shouldn't really matter, right?

If possible, can this be done by only changing the Pins for the Gear Motor to the respective ones on the Octopus or would this also require modifications to the code?

LostFool, 2023-07-04:
I'd love to know if you tried this and got results.

Durahl, 2023-07-05:
[Happy Hare V3](https://github.com/moggieuk/ERCF-Software-V3) - which I've been running on my Setup for a while now ( and which I can **_HIGHLY_** recommend! ) - now seems to be running this very feature in their latest 1.3.0 Release but I have so far not tested it with it being enabled in most parts because the issue from my OP has been resolved by ditching the ERCP for a DIY Solution that does away with a lot of friction causing elements.