EtteGit/EnragedRabbitProject #149: Filament runout operation?

amv101, 2022-09-06:
Quick question. Does the clog detection mode need to be enabled for the Klipper filament runout feature to work? Can someone elaborate how it should work? Thanks in advance!

Jaans, 2022-09-08:
I too would like more info 🙂

Jaans, 2022-09-12:
Yes, that is the ERCF equivalent of filament runout or jam detection. The ERCF Documentation manual page 121 details how to set it up. Some seem to have success and other less so. 

I've recently set it up and had to tweak the ```detection_length``` and the ```variable_final_load_check_threshold``` settings for my scenario and so far are having the desired outcome, including when the filament loads into the extruder but doesn't go into the cooling tube (an actual clog).