EtteGit/EnragedRabbitProject #160: Documentation: missing info on rod length for different numbers of filament blocks

drelich, 2022-10-09:
Going through the documentation, I can see info on how long the wires need to be for a 3, 6 and 9 block version of the ERCF. I cannot find the same kind of info on the length of the rods.

olivbd, 2022-10-18:
On page 18, you can find typical wire lengths for the selector motor.

[image]

And on the next page, you can find for sensor and encoder wires.

[image]

drelich, 2022-10-18:
Yes, I know, and I acknowledged that. I was asking about the rod lengths, not the wires :) Cheers.

olivbd, 2022-10-18:
Rod lengths are mentioned in the BOM

https://docs.google.com/spreadsheets/d/1A_qeNeaVprh_iPbvLcLjvJ2o2NfCzjuJam3K3RkWW0w

You have to enter the # number of channels you want and it calculates the lengths for you ;-)

drelich, 2022-10-18:
Good point. Thanks. BTW: You have to create a copy of that BOM in order to change that number of channels. By default, that BOM opens with View permissions only :)

olivbd, 2022-10-18:
You're welcome. Yes, you have to create a copy in your own Google Drive storage.

Nevertheless, I agree with you, rods lengths could be mentioned in the documentation. And wires also in the BOM.