EtteGit/EnragedRabbitProject #167: Encoder cart tolerance/clearance issue

SkiBikePrint, 2022-10-25:
[image]

Design tolerance between bearing and printed part is only 0.2mm per the circled area in the attached picture.  The printed part was too tight and prohibiting the bearing from turning.  Expected print tolerances create likely interference.  The part in the picture has been filed to open up the clearance. 

There does not seem to be a design requirement to have the spacing so tight.  I would suggest opening up the clearance to at least 0.5mm or a minimum radius of 5.5mm for the bearing pocket.

Thanks, Glenn
[image]

jwhite, 2022-11-07:
@SkiBikePrint Can you submit a pull request with your proposed changes?

SkiBikePrint, 2022-11-07:
Hi Jeremy,

I submitted a pull request.  I have never done one before so don't know if I did it right.  Although I created edited STL's I don't know how to attach them so I included them in this email.  Let me know if there is more I need to do.

Thanks,

Glenn