EtteGit/EnragedRabbitProject #168: "a filament sensor system located below the extruder gears" vs Direct Drive Extruder on Printer?

theschles, 2022-11-05:
Hi there, I have a Sovol SV-01 running Klipper, which has a direct-drive extruder -- and no room for a filament sensor between the printer's extruder gears and the hot-end.  

When you say "The filament sensor : a filament sensor system located below the extruder gears", to which extruder are you referring?

* The printer's own extruder?
  * ...which would mean that I can't get the ERCF for my printer...
-- or --
* The extruder on the ERCF?
  * ...which would mean my printer would work: Spools -> ERCF -> Filament Sensor -> Direct Drive Extruder -> Hot End

Thanks for the clarification in advance!

RomRider, 2022-11-05:
It's supposed to be after the toolhead extruder, however it's not required anymore if you use the "no_toolhead_sensor" branch

theschles, 2022-11-06:
> It's supposed to be after the toolhead extruder, however it's not required anymore if you use the "no_toolhead_sensor" branch

Apologies, I'm still learning the terminology...when you say "toolhead extruder", you're referring to the 3D printer's extruder, correct?

opiswahn, 2022-11-21:
@theschles The toolhead is usually the thing your hotend is in, if you have DirectDrive then the Extruder is included. The whole assembly is called a tool (hence the name Toolchanger, eg. from E3D)

closed 2022-11-21 by theschles