EtteGit/EnragedRabbitProject #177: PLA on ERCF?

BoredFishRE, 2022-12-01:
I really can't do ABS, so for anyone that did print in PLA, is there anything I should do for PLA? For the test print, scaling it down by 5% seems to make the heat set insert fit properly, though I don't know if I should do that since I don't know if it'll make any clearance issues.

RomRider, 2022-12-01:
I printed in PETG, no difference and no modifications. 5% is A LOT. Either you don't have the right inserts or your printer is not calibrated.
Use a caliper instead and compare against the CAD.

BoredFishRE, 2022-12-01:
> I printed in PETG, no difference and no modifications. 5% is A LOT. Either you don't have the right inserts or your printer is not calibrated. Use a caliper instead and compare against the CAD.

ah. I just checked cad, it's 4.7mm and the CNC kitchen heat set inserts i'm using are 4.6mm in dia, so that may be my problem

RomRider, 2022-12-01:
Heat inserts are M3xD5xL4, you can find them on AliExpress

BoredFishRE, 2022-12-01:
I got a kit from dfh, so I think I'll be Good

closed 2022-12-01 by BoredFishRE