EtteGit/EnragedRabbitProject #204: tolerances of .stl parts good for resin printing?

o-lampe, 2023-10-16:
Often, parts designed for FDM printers come with a *fudge factor* for more precise inner diameters and bridging.
If I print ERCF parts with a resin printer, will they fit together or are too loose or tight?
THX
O-Lampe

ammoniak, 2023-10-26:
There is a calibration part, The documentation/manual explains the features and what you need to messure/test. If the clibration part works, the rest should too.

o-lampe, 2023-11-08:
The calibration part worked fine, but there are so many M3 heat-set inserts, which won't work with resin-printed parts.

Kadah, 2023-12-27:
> The calibration part worked fine, but there are so many M3 heat-set inserts, which won't work with resin-printed parts.

You could try over-sizing those holes and setting the inserts with resin. 

I had a quick google and CNC Kitchen did a vid on this very topic last year.
https://www.youtube.com/watch?v=4nbvU2kGzxc